unl-nimbus-lab/Freyja #16: Definition of Current State

hildebrandt-carl, 2022-02-17:
Hi I was curious what `current_state` held. I know its a 13 cell array.

I have figured out the following:
* `current_state[0]` = x position
* `current_state[1]` = y position
* `current_state[2]` = z position

I assume the following
* `current_state[3]` = x velocity
* `current_state[4]` = y velocity
* `current_state[5]` = z velocity

I am not sure of the rest. I know that some controllers track acceleration? I am also not sure where yaw would be.

Thanks in advance :)

ajshank, 2022-02-17:
Does this help : https://github.com/unl-nimbus-lab/Freyja/wiki/Setup-motion-capture ?

hildebrandt-carl, 2022-02-17:
That definitionally does shed some light.

So now I know:
* `current_state[0]` = x position
* `current_state[1]` = y position
* `current_state[2]` = z position
* `current_state[3]` = x velocity
* `current_state[4]` = y velocity
* `current_state[5]` = z velocity
* `current_state[6]` = **unknown**
* `current_state[7]` = **unknown**
* `current_state[8]` = yaw
* `current_state[9]` = **unknown**
* `current_state[10]` = **unknown**
* `current_state[11]` = **unknown**
* `current_state[12]` = vicon inter-message interval

There are still 5 variables which I cant figure out :(  Based purely the fact that there are 3 unknowns in a row I assume:

* `current_state[9]` = x acceleration
* `current_state[10]` = y acceleration
* `current_state[11]` = z acceleration

But you know what they say about assumptions.

ajshank, 2022-02-18:
The wiki perhaps needs a page for this sort of thing, but your answer is here: https://github.com/unl-nimbus-lab/Freyja/blob/master/freyja_msgs/msg/state_manager/CurrentState.msg
The topic `current_state` is the general interface for the controller. When flying outdoors, this data is populated by GPS data (presumably from the autopilot) -- the topic remains the same, it is just updated/combined from various sources (depending upon the `state_source` launch parameter). Yet another data source, for instance, is odometry produced by onboard cameras (monocular or Realsense-type).

hildebrandt-carl, 2022-02-18:
Thank you! Thats exactly what I was looking for :)

* `current_state[0]` = x position
* `current_state[1] `= y position
* `current_state[2]` = z position
* `current_state[3]` = x velocity
* `current_state[4]` = y velocity
* `current_state[5]` = z velocity
* `current_state[6]` = roll
* `current_state[7]` = pitch
* `current_state[8]` = yaw
* `current_state[9]` = roll rate
* `current_state[10]` = pitch rate
* `current_state[11]` = yaw rate
* `current_state[12]` = inter-messsage interval

closed 2022-02-18 by hildebrandt-carl