unl-nimbus-lab/Freyja #19: Build Issue with Ardupilot computer_ctrl

pfletche, 2022-03-11:
Hi @ajshank, 

I've been unable to build (with ./waf) the computer_ctrl version of Ardupilot ([maintained here](https://github.com/ajshank/ArduPilot)) for SITL. It also doesn't seem to build for other boards. It may be that it's inconsistent with the most recent version of Ardupilot? I tried dropping the computer_ctrl files into the most recent version of Ardupilot but there are also inconsistencies. 

Have you been able to build this [version](https://github.com/ajshank/ArduPilot) of Ardupilot with computer-ctrl? If not, is Freyja supported for 'Guided' mode in Ardupilot or 'Offboard' in PX4?

Hope you are well!

ajshank, 2022-03-14:
@pfletche Thanks for testing and reporting this. I'm well; hope you're too :)

I presume that you've cloned the entire repo (with `--recurse-submodules`)? I don't have a complete toolchain installed on my machine at the moment, but if you post a specific error, I can try to replicate it. It is also possible that a new version of Ubuntu has a new version of `gcc` or `libc` that no longer recognises some symbols. Are the precompiled binaries on Box/OneDrive for a different board than yours?

Freyja will work with `Guided-NoGPS` mode (since Ardupilot 4.1+, some of my changes are now available in the main repository), and PX4's `OFFBOARD` should also work (very much untested). However, please keep in mind that there is a small segment of code in `apm_handler` package that looks for the _string_ `COMP_CTRL`, and makes a couple of service calls. This behaviour will no longer work, but is not essential to core flight control (in general, none of Freyja's internal functions depend on anything external). If this is the route you prefer, there are a few minor adjustments that will need to be made. Please let me know.

pfletche, 2022-03-15:
Thanks, Ajay! Yep, I did clone the entire repo. I pasted the specific error below. Looks like there are too many args in a function in AP_Generator_RichenPower.cpp. 

I'm now trying Guided_NoGPS. Looks like I need to change the G2_OPTIONS parameter, where does that param live exactly? I tried changing GUID_OPTIONS in ardupilot/ArduCopter/Parameters.cpp to 3: interpret thrust as thrust, but that hasn't worked. 

----------------------
$ ./waf copter
Waf: Entering directory `/home/paul/Ardupilot_Ajay_Original/ardupilot/build/sitl'
[3/6] Compiling libraries/AP_Scripting/generator/src/main.c
[4/6] Processing modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
[5/6] Creating build/sitl/ap_version.h
Validation skipped for /home/paul/Ardupilot_Ajay_Original/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml.
Parsing /home/paul/Ardupilot_Ajay_Original/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Validation skipped for /home/paul/Ardupilot_Ajay_Original/ardupilot/modules/mavlink/message_definitions/v1.0/common.xml.
Parsing /home/paul/Ardupilot_Ajay_Original/ardupilot/modules/mavlink/message_definitions/v1.0/common.xml
Validation skipped for /home/paul/Ardupilot_Ajay_Original/ardupilot/modules/mavlink/message_definitions/v1.0/uAvionix.xml.
Parsing /home/paul/Ardupilot_Ajay_Original/ardupilot/modules/mavlink/message_definitions/v1.0/uAvionix.xml
Validation skipped for /home/paul/Ardupilot_Ajay_Original/ardupilot/modules/mavlink/message_definitions/v1.0/icarous.xml.
Parsing /home/paul/Ardupilot_Ajay_Original/ardupilot/modules/mavlink/message_definitions/v1.0/icarous.xml
Merged enum MAV_CMD
Found 227 MAVLink message types in 4 XML files
[6/6] Processing /home/paul/Ardupilot_Ajay_Original/ardupilot/build/sitl/libraries/AP_Scripting/lua_generated_bindings.cpp,/home/paul/Ardupilot_Ajay_Original/ardupilot/build/sitl/libraries/AP_Scripting/lua_generated_bindings.h: libraries/AP_Scripting/generator/description/bindings.desc build/sitl/gen-bindings -> build/sitl/libraries/AP_Scripting/lua_generated_bindings.cpp build/sitl/libraries/AP_Scripting/lua_generated_bindings.h
Generating C implementation in directory /home/paul/Ardupilot_Ajay_Original/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v2.0/ardupilotmega
Generating C implementation in directory /home/paul/Ardupilot_Ajay_Original/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v2.0/common
Generating C implementation in directory /home/paul/Ardupilot_Ajay_Original/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v2.0/uAvionix
Generating C implementation in directory /home/paul/Ardupilot_Ajay_Original/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v2.0/icarous
Copying fixed headers for protocol 2.0 to /home/paul/Ardupilot_Ajay_Original/ardupilot/build/sitl/libraries/GCS_MAVLink/include/mavlink/v2.0
[  7/709] Compiling libraries/AC_AttitudeControl/AC_AttitudeControl_Sub.cpp
[  8/709] Compiling libraries/AC_AttitudeControl/AC_AttitudeControl_Heli.cpp
[  9/709] Compiling libraries/AC_AttitudeControl/ControlMonitor.cpp
[ 10/709] Compiling libraries/AC_AttitudeControl/AC_AttitudeControl_Multi.cpp
[ 11/709] Compiling libraries/AC_AttitudeControl/AC_PosControl_Sub.cpp
[ 12/709] Compiling libraries/AC_AutoTune/AC_AutoTune.cpp
[ 13/709] Compiling libraries/AC_Autorotation/AC_Autorotation.cpp
[ 14/709] Compiling libraries/AC_Avoidance/AP_OADatabase.cpp
[ 15/709] Compiling libraries/AC_Avoidance/AP_OADijkstra.cpp
[ 16/709] Compiling libraries/AC_Avoidance/AP_OABendyRuler.cpp
[ 17/709] Compiling libraries/AC_Avoidance/AP_OAVisGraph.cpp
[ 18/709] Compiling libraries/AC_Fence/AC_PolyFence_loader.cpp
[ 19/709] Compiling libraries/AC_InputManager/AC_InputManager.cpp
[ 20/709] Compiling libraries/AC_InputManager/AC_InputManager_Heli.cpp
[ 21/709] Compiling libraries/AC_PID/AC_P.cpp
[ 22/709] Compiling libraries/AC_PID/AC_HELI_PID.cpp
[ 23/709] Compiling libraries/AC_PID/AC_PI.cpp
[ 24/709] Compiling libraries/AC_PID/AC_PID.cpp
[ 25/709] Compiling libraries/AC_PID/AC_PID_2D.cpp
[ 26/709] Compiling libraries/AC_PID/AC_PI_2D.cpp
[ 27/709] Compiling libraries/AC_PrecLand/AC_PrecLand_IRLock.cpp
[ 28/709] Compiling libraries/AC_PrecLand/AC_PrecLand_SITL_Gazebo.cpp
[ 29/709] Compiling libraries/AC_PrecLand/PosVelEKF.cpp
[ 30/709] Compiling libraries/AC_PrecLand/AC_PrecLand_SITL.cpp
[ 31/709] Compiling libraries/AC_PrecLand/AC_PrecLand.cpp
[ 32/709] Compiling libraries/AC_PrecLand/AC_PrecLand_Companion.cpp
[ 33/709] Compiling libraries/AC_Sprayer/AC_Sprayer.cpp
[ 34/709] Compiling libraries/AC_WPNav/AC_Circle.cpp
[ 35/709] Compiling libraries/AC_WPNav/AC_Loiter.cpp
[ 36/709] Compiling libraries/AC_WPNav/AC_WPNav.cpp
[ 37/709] Compiling libraries/AC_WPNav/AC_WPNav_OA.cpp
[ 38/709] Compiling libraries/AP_ADC/AP_ADC_ADS1115.cpp
[ 39/709] Compiling libraries/AP_AHRS/AP_AHRS_DCM.cpp
[ 40/709] Compiling libraries/AP_AHRS/AP_AHRS_View.cpp
[ 41/709] Compiling libraries/AP_AccelCal/AP_AccelCal.cpp
[ 42/709] Compiling libraries/AP_AccelCal/AccelCalibrator.cpp
[ 43/709] Compiling libraries/AP_AdvancedFailsafe/AP_AdvancedFailsafe.cpp
[ 44/709] Compiling libraries/AP_Airspeed/AP_Airspeed_UAVCAN.cpp
[ 45/709] Compiling libraries/AP_Airspeed/AP_Airspeed_MS5525.cpp
[ 46/709] Compiling libraries/AP_Airspeed/AP_Airspeed_analog.cpp
[ 47/709] Compiling libraries/AP_Airspeed/Airspeed_Calibration.cpp
[ 48/709] Compiling libraries/AP_Airspeed/AP_Airspeed_Backend.cpp
[... cut ...]
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:34: warning: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ specified bound 65 equals destination size [-Wstringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Waf: Leaving directory `/home/paul/Ardupilot_Ajay_Original/ardupilot/build/sitl'
Build failed
 -> task in 'objs/AP_Generator' failed (exit status 1): 
	{task 140262601182896: cxx AP_Generator_RichenPower.cpp -> AP_Generator_RichenPower.cpp.0.o}
 (run with -v to display more information)

radall1, 2023-07-25:
We're also trying to use Freyja with a recent version of Ardupilot with GUIDED_NoGPS (and adjusting G2_Options) but it didn't work. What was the solution (if any) for you?

johnthelzer, 2024-05-30:
> @pfletche Thanks for testing and reporting this. I'm well; hope you're too :)
> 
> I presume that you've cloned the entire repo (with `--recurse-submodules`)? I don't have a complete toolchain installed on my machine at the moment, but if you post a specific error, I can try to replicate it. It is also possible that a new version of Ubuntu has a new version of `gcc` or `libc` that no longer recognises some symbols. ~Are the precompiled binaries on Box/OneDrive for a different board than yours?~ Sorry, I just saw that you're looking for SITL.
> 
> Freyja will work with `Guided-NoGPS` mode (since Ardupilot 4.1+, some of my changes are now available in the main repository), and PX4's `OFFBOARD` should also work (very much untested). However, please keep in mind that there is a small segment of code in `apm_handler` package that looks for the _string_ `COMP_CTRL`, and makes a couple of service calls. This behaviour will no longer work, but is not essential to core flight control (in general, none of Freyja's internal functions depend on anything external). If this is the route you prefer, there are a few minor adjustments that will need to be made. Please let me know, this should be doable.

@ajshank I just found this thread- I'm using Freyja with Arducopter v 4.5.1 on a blue cube. I'm also using Guided_noGPS mode. Since this is a later version than Arducopter v 4.1, does it require any custom firmware package, or should it work as is?

Thanks for your help,
John

ajshank, 2024-08-07:
Hi @johnthelzer,
Unless you have a strong dependency on 4.5, I'd recommend building this 4.4 firmware: https://github.com/ajshank/ardupilot/tree/freyja4.4
This forks from 4.4 and adds a yaw-rate control functionality that was removed from ArduCopter after 4.2. No other base functions are modified. Please feel free to start a thread on Discussions if you'd like and we can move the conversation over there.

johnthelzer, 2024-08-09:
@ajshank Thank you! That is very helpful.