unl-nimbus-lab/Freyja #25: Sr2 publish rates

johnthelzer, 2024-05-31:
@ajshank Hi Ajay,
The Freyja Autopilot instructions say (as per attached screenshot):
- "sr2: change all publish rates to zeros, except for EXTRA3 (battery info, system state) which can be set to 1Hz. "
However, when I did this, mavros stopped transmitting GPS, IMU, and odometry data, which caused the current_state topic and state manager node to not work.  Is the sr2 publish rates recommendation still valid for later versions of arducopter (4.1+), or is it correct to set publish rates to non-zero?

Thanks,
John

[image]

ajshank, 2024-08-07:
@johnthelzer Yes, this recommendation is correct; however, this is geared mostly towards indoor flights (controlled via radio modules), given that you have an alternative positioning source offboard (like mocap). Since you are interested in GPS & odometry messages, I'm guessing this is not an indoor setup - in which case, I'd recommend setting the rates for GPS and IMU messages to what you need them to be (with the understanding that these are often big packets to stream). If you only have one radio, then it is going to have to handle all of that, plus the controller's signals back to the aircraft. Hopefully you know your radio limits :)

If you describe what your setup is, perhaps my recommendation might be more targeted. And it might be worth moving this to Discussions as opposed to Issues.

Apologies for not seeing this sooner -- something's wrong with my notification settings.

johnthelzer, 2024-08-09:
@ajshank All good, thanks for replying! I ended up setting GPS and odometry to normal levels and it worked; I'm using a raspi hardwired to the FMU, so that provides more bandwidth than a radio would. I actually have moved away from field testing for a while as I write a paper, but I'll open a discussion next time.

Thanks again!

closed 2025-04-07 by ajshank