unl-nimbus-lab/Freyja #28: Hover mode overshoot

johnthelzer, 2025-04-03:
Hi Ajay,
I've been having some trouble running Freyja on a DJI F550 drone outside. The drone flies great in alt-hold, loiter, and other flight modes, but there's something going on with the commands sent by Freyja. I'm using your forked Freyja ArduCopter 4.4 and all the suggested settings from the Freyja wiki. Additionally, I was able to get Freyja working in ArduPilot's software-in-the-loop simulation, so I think it's an issue specific to this implementation.

System parameters:
- Drone mass = 2.39 kg
- Battery: 4s Lipo (14.8V)
- APM version = 4.4.1 (forked Freyja)
- Drone = DJI F550
- ESCs and motors: DJI 430 Lite ESCs, 2312 motors, 9 inch props

I determined the correct thrust scaler (I think) by manually changing the commands in mavros_translate.cpp to 0 degrees roll and pitch and a thrust of m*g/thrust_scaler and adjusting the thrust scaler until the drone hovered. The thrust scaler turned out to be around 130, which seems reasonable. 

However, when I ran Freyja using this thrust scaler and the hover reference point of 1 meter (pz_ref = -1.0), it exhibited the behavior shown in the attached video. In the video, I started off in loiter mode, then switched to guided_nogps. As soon as I switched to guided_nogps, the drone rapidly overshot the setpoint by a few meters, behaved erratically, then rapidly fell to the ground. This behavior has happened with many iterations of the thrust scaler and reference point.  Looking at the topics published, it looks like the reference state, current state, and commands were being sent correctly. 

Do you have any insight into what could be causing this problem? Do I need to recalculate the K matrix in the LQR controller?

[https://uofnelincoln-my.sharepoint.com/:v:/g/personal/jhelzer3_unl_edu/EcdQrSSD56dEg_PH8xV4sn8Bzg34xPlzf-tEZw3dYJHhkw](url)

Thanks,
John

ajshank, 2025-04-07:
Not sure I follow based simply on your description -- and I cannot access the linked video. If you have a log file with some relevant topics (current_state, reference_state and controller_debug, at least), it may be possible to debug this better.

johnthelzer, 2025-04-15:
Hi Ajay,
Sorry, that wasn't a very helpful question. I think a better question I have about Freyja is about the thrust scaler and shape of the thrust curve. 

I'm trying to use a modification of Freyja where I replace just the controller with a custom controller, and to test that the inputs are being transmitted properly to the drone, I am doing tests where I command a specific input through the "mavros_translate" node. In this case, I want the drone to "hover" (roll, pitch = 0, yaw = desired direction, thrust = drone_mass*g). I successfully got the drone to "hover" at a certain mass and thrust scaler (2.254 kg and 110), but when I increased the mass with the same thrust scaler, the drone did not have enough thrust to maintain altitude and fell instead of hovering. I validated that the desired throttle was sent correctly to mavros (in the /mavros/setpoint_raw/attitude topic), so I'm thinking the problem must be with how the throttle is turned into thrust.

Right now I'm using the default mot_thst_expo of 0.65- this is based just on prop size and I haven't validated it with a thrust stand yet. My understanding of how Mavros accepts throttle in Guided_nogps with GUID_OPTIONS = 10 is that the thrust scale from 0 to 1 is based on the PWM values set in ArduCopter. In my case, I calibrated the ESCs and set a range of 1105-1925. I'm not 100% sure this is the correct range, because I set it based on the limits of the RC controller, but it seems to make the copter fly fine in Loiter and Althold modes. 

If my hypothesis is correct, the throttle to thrust curve is nonlinear and the mot_thst_expo is correct or close to correct. This would mean that the mapping between throttle PWM values and the resulting thrust force is not a linear relationship.

If the throttle-thrust mapping were linear, then the thrust scaler would make sense. However, did you account for nonlinear thrust curves with Freyja's thrust scaler, or is that just something that the controller compensates with?

Thanks,
John

johnthelzer, 2025-06-15:
I realized a while after asking this that I was wrong- the scaled thrust is truly thrust, and the thrust scaler method you used is accurate. I think the problems I was having were due to the flipped z-direction velocity from issue #29, which caused Freyja to freak out because it was receiving an inverted vertical velocity! The linear thrust relationship you assumed works great, especially with a controller that can provide more or less thrust as needed and make up the difference.

Freyja works great now! It's really a life-saver, especially handling all these little implementation details like scaled thrust that would take forever to figure out otherwise.

closed 2025-06-15 by johnthelzer

ajshank, 2025-06-17:
Glad to hear! 
Please feel to start new Discussions -- in some cases perhaps more apt than Issues. Would be great to see successful cases as well, so the discussion can be more about improvements, quality-of-life fixes etc. No pressure, of course!