unl-nimbus-lab/Freyja #29: Velocity Frame and Direction

johnthelzer, 2025-05-27:
Hi Ajay,

I have 2 questions about Freyja's implementation; specifically, the state manager node.

1. **Velocity frame convention:**

The /mavros/global_position/local topic says the child frame is base_link; this is the frame that the twist (linear and angular velocity) should be in. However, the state callback for the topic uses the velocity without any frame conversion as if it was in the world-fixed map or odom frame. This seems to work fine, but I'm curious why. I looked a bit in the config files (apm_config_nimbus.yaml) and the global position child frame is listed as base_link there too.  Is there something in your Ardupilot fork that switches the frame convention for global position?

2. **Vertical velocity difference**

I was struggling for a while to implement Freyja at all until I noticed the sign on vz (obtained from /mavros/global_position/local) is the opposite of what it should be. Vz from mavros should be in the upward direction so vd, in the downward direction, is found with vd = -vz. However, the vz I'm getting seems to be in the _downward_ direction. When I changed the sign in gpsodom_callback to vd = vz (instead of vd = -vz), the controller worked. I can't find any explanation for why mavros would provide the wrong sign for vertical velocity, though. 

I encountered this issue on more than one Orange Cube Plus flight controller and in ArduPilot's SITL simulation, all running your ArduCopter 4.4 Freyja fork. This makes me think it's definitely not an issue with hardware. Do you know what might cause this bug?

Thanks,
John

ajshank, 2025-06-14:
Sorry for this confusion - the problem with direction of vertical velocity has been a long-running one, and I've had to fight with it frequently. Mavros does some internal conversions (to meet ROS conventions), and I am not sure I'm fully convinced that all vehicle topics are represented correctly. It would help to know which mavros version you are using.

As a quick note, outputs of state manager are supposed to be in a fixed-frame (world, map etc). Because of the way the mapping is done in subscriber callbacks, it makes it necessary to subscribe to a topic that is already in a fixed-frame (but could be ENU, for instance). There is no additional frame conversion that is done in Freyja or in AP; if you observe that a mavros topic does not look like it is in base_link, then I'd imagine that the data is correct but the naming is not.

johnthelzer, 2025-06-16:
That's ok- other than the direction inverted, all the other data received seems to be fine. I'm using Mavros version 1.17.0. The data from global_position/local is all correct other than the vertical velocity.

ajshank, 2025-06-17:
Great. Hope it is ok to close this then? Feel free to reopen if needed.

closed 2025-06-17 by ajshank